xibyte/jsketcher #3: How to install dependencies?

mmiscool, 2016-12-15:
Hello,

Sorry to bother you but how can I go about installing this on my own server. I have cloned the git in to a directory on my server but when i visit the page receive blank page with resources it can't load.

xibyte, 2016-12-15:
Hello,

You need to install node.js and npm and then do from the app directory

**$ npm install** 

It installs the dependencies locally inside the app dir.

Since it's a webpack app you can either build a bundle or just run the webpack dev-server(would the easiest way)

**$ npm start** #from the root of the app dir

and then the app will be available at http://localhost:3000

I also have plans to check the bundled application(with all the dependencies) in the git repository in future

xibyte, 2016-12-18:
Added these instructions to the README

closed 2016-12-18 by xibyte

ThorsenRune, 2020-03-30:
Sorry but I don't know node.js. It HAS to be installed? It cannot run without?.
So it's not simple to install it on a hosted service!?.