xibyte/jsketcher #38: Cannot launch - 'cannot find handlebars'

cyplo, 2016-12-26:
Hi !
Thank you for this project - love it !
I tried lauching locally on Fedora 25, npm 3.10, node 6.9.1.
`npm install` gives:

```
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.0.15: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN handlebars-loader@1.4.0 requires a peer of handlebars@>= 1.3.0 < 5 but none was installed.
```

If I do `npm start` then it goes like this:
```
> jsketcher@0.1.0 start /home/cyryl/dev/jsketcher
> webpack-dev-server --config webpack.config.dev.js --content-base web/ --port 3000

 http://localhost:3000/webpack-dev-server/
webpack result is served from /static/
content is served from /home/cyryl/dev/jsketcher/web
```

If I'll try to access the URL given, however, I get:
```
WARNING in Cannot find module 'handlebars'
 @ ./web/app/3d/ui/tmpl ^\.\/.*\.html$

WARNING in Cannot find module 'handlebars'
 @ ./web/app/3d/ui/tmpl ^\.\/.*\.html$

WARNING in Cannot find module 'handlebars'
 @ ./web/app/3d/ui/tmpl ^\.\/.*\.html$

ERROR in Cannot find module 'handlebars'
 @ ./web/test/runner.js 5:16-48
webpack: bundle is now VALID.
```
[image]

I am happy to help here - please let me know if I'm doing something wrong or if you need more information.
Thank you !

mmiscool, 2016-12-26:
Try to do the following
 npm install handlebars

Looks like this is going to have to be added to the dependency list for the package so it is handled with the normal install.

cyplo, 2016-12-26:
It indeed helped, the application view is showing now, thank you !
It seems that nothing is clickable though, will investigate further - I don't want to pollute this issue so will open another one if I confirm. Thank you !

xibyte, 2016-12-26:
Thanks for reporting it @cyplo. I just added explicit dependency to package.json(667c5d37) Now it should work OOTB.

closed 2016-12-26 by xibyte