xibyte/jsketcher #39: Feature request: Part level expressions.

mmiscool, 2016-12-27:
This is to make the model able to be edited by changing a few values.

Establish variable names.

Ability to use these variable names for dimensions in the sketcher and the dialogs for features. 

Ability to use complex math such as +,-,*,/,(,),^,ect.

List the variables in a window that can be easy edited.

Each dimension established or value entered in to feature dialog should also create an expression that can be seen and edited in the expression dialog.

Evil-Spirit, 2018-12-17:
@mmiscool You're welcome to NoteCAD where such feature planned to implement.

mmiscool, 2018-12-17:
jsketcher just implemented part level expressions. These are not yet consumed in to the individual sketches yet but work for all the feature dialogs. On a side note. Is that based on solvespace resolver there?

Evil-Spirit, 2018-12-18:
This is based on SolveSpace's solver ideas, but all the code was written from scratch (I have recorded videos to proove that).

Arredatore, 2020-03-25:
Meanwhile, a bit offtopic, I have done my simple online editor on pure JS, at that time I was looking for a platform to do it simpler. Based on what I've learned it is more interesting for me now to have a library, kind of "Lodash for geometry", with geometry solver, intersection solver, booleans etc.

mmiscool, 2022-08-07:
This issue has been solved for a very long time. Seem to have forgotten to close it.

closed 2022-08-07 by mmiscool