xibyte/jsketcher #44: jsketcher's internals deserves to be a project on it's own - do it

dkipp, 2017-03-30:
I’d like to encourage you to split your truly excellent work in two separate projects.

1. jsketcher as the UI
2. javascript based solid geometric modelling kernel

What you have develop looks awesome and separating internals into an UI independent JavaScript solid geometric modelling kernel would open tremendous possibilities and for sure attract other developers to support its development further.

I’m sure you considered this already, so what are your thoughts?
Daniel

ShreyasWao, 2017-03-30:
I agree wholeheartedly with Daniel ! The usefulness of having a web-based parametric modeller is that you can finally give the end user some freedom to make changes to the 3D model and make it useful for them (and trust me a threejs based parametric modeller is way better than the ones on platforms like thingiverse) . But sometimes you might want to do the final model generation on the server side ( for security reasons and to protect your design ip).  Project part 2 will be especially useful for this purpose. 

Do let me know if you are keen to pursue this, I am willing to help out whatever way possible.

dkipp, 2017-03-30:
My motivation for asking this is more of a basic need.

WebGL is great and it gives us a very powerful possibility to use the GPU from within the browser - but it is just that. It provides nothing for 3d. No cameras, no lights, no geometries – everything needs to be provided by the developer. 

Willing to use WebGL without a framework. Every developer is faced with the same task. To find a  Math and Matrix library to calculate everything and get some polygons to display.

While it is easy to find a math and matrix library, it is very hard to find a generic JavaScript solution for the basic geometry needs and client side shape generation.

I’d like to see a JavaScript based, app independent geometry kernel, capable of programmatically building and modifying shapes. For me, jsketcher looks like this but is currently bound to an UI.

The Library is ideally not based on polygons, but obviously capable of triangulating the created solid to be used with WebGL. Potentially allowing the export to other 3D/CAD file formats supporting surface definition superior to polygons.

The part export is by no means a feature request, but with an library or kernel open to developers to whatever JavaScript/WebGL project there are up to, I see features like this coming in by other contributors. Or even next-gen frameworks building upon that kernel.

xibyte, 2017-03-31:
Good point. That makes sense. I am currently working on a BREP engine and it will give us good logical separation the core(engine) from the UI. Once brep is done it will be possible to split the codebase for two separate projects as you mentioned. BREP core isn't based on polygons but instead based on surfaces and their bounds representation providing an API to triangulate it. A lot of work related to BREP is already in master branch but there still some work to get done.

dav-m85, 2017-11-05:
@dkipp have you look into [csg.js](https://github.com/jscad/csg.js) ? the current work on splitting it from the openjscad ui has gone far, you can run now js code to build shapes in node with a constructive solid geometry approach, [for examples check here](https://en.wikibooks.org/wiki/OpenJSCAD_User_Guide#Intersection).

dumblob, 2019-11-28:
@xibyte how does the BREP work look like? Do you need any help?