xibyte/jsketcher #46: Feature Request: Refactor code to reusable components for use in other app frameworks

bradodarb, 2017-06-22:
**1. This is amazing work, thank you so much for sharing! Wow!
2. Not sure if you'd be at all interested but I think this body of work deserves to be split up and shared:**

I'd like to contribute to this project by pulling apart the components so that they may be used in various scenarios such as an NG2 app, React app, existing custom app framework, etc...  A lot of the code is clean and modular already, I'd just like to further that effort and break things into semi-self contained modules such as:

- 2d sketcher
- 3d sketcher
- Separate geometry data and utils from presentation objects

I've already started pulling things apart and I'm trying to get a 2d sketcher module built.
It will take a while for me to grok some of your algorithms and guidance from you would be greatly appreciated.

xibyte, 2017-06-28:
I am actually actively working on merging the sketcher right inside 3D view which requires some level of separation of the sketcher from the current 2D UI in my private branch.  I'll able to merge it to master soon. 

The guidance for you to bundle a sketcher as a separate module would be to start with the class viewer2d/Viewer and try to separate it and see what dependencies it pulls out. After that to find all the global usages of jquery and try to find a way to abstract them out. Hope it helps!

bradodarb, 2017-06-28:
I've started a project to break the 2d things apart.  I started it using typescript to try and help me discover dependencies, but it could be easily migrated to pure es6/7.  Also, I have it under an NG app for development convenience, but it will be broken into it's own repo without the angular dependencies once initial migration is complete.

Very early stages, let me know what you think @xibyte 
https://github.com/bradodarb/m4-sketcher