xibyte/jsketcher #51: triangulation do not support nurbs-curve directly

iamnotstone, 2017-09-07:
I don't know if this is a feature loosing or something unproper. In my use case, I'd like to triangulate a closed face enclosed by nurbs-curve directly. After taking a deep insight your excellent code, I found you never do this, but by converting every thing to polygon and then calling the libtess's function.
May be we are just on different use case, I'd like to improve your code in '{project}/web/app/3d/triangulation.js', and to make the 'TriangulateFace' function to support nurbs-curve. If you think it also useful for your project, I'd like to be a collaborator for you.
And as a code reader, I think it's better for this project to split the 'brep' and 'brep-3d' part to another project. And make it to be a modeling library project for all of the modelling project on web.

xibyte, 2017-09-08:
Yes you are right. It would be really helpful, thanks for your help. 

I think we can approach the problem following way, but I am not sure and may be it's too naive, correct me if I am wrong:

1. Divide all bounding curves of the face to some ranges taking into account their curvatures. Each range should represent approximately a flat part of the curve.  I guess that exactly what method *verb.Tess.rationalCurveAdaptiveSample* does.
2. Then transfer all the ranges to the parametric space of the face. 
3. We'll get some sort of '2D polygons' on the parametric space which we can triangulate. Then do that triangulation on the face's parametric space. We'll get the triangles but it's not enough. The problem is that the such triangles don't take into account the curvature of the enclosed area.
4. To solve this problem we need to evaluate derivative at the middle of the triangle and at the each triangle's vertices. If they the derivatives are about the same it is a good triangle and covers not curving area and it should go to the result set of the triangulation. Otherwise we split triangle from to three sub-triangles(using 3 vertices and the center) and repeat step 4 for the sub-triangles.
5. However, if a triangle goes through any knot in any direction(u or v) we also need to split the triangle by the knots line before doing the curvature analysis. Otherwise such triangle can cover a "local hump" which we never know about.

PS. Actually the whole nurbs infrastructure is in very bad and awful shape and there are a lot of weird code to workaround the limitations of the current boolean algorithm(which currently can work only on polyhedra). So I am planing to rewrite everything related to the NURBSes or at least the parts related to  burbs solids construction and performing the boolean operations as soon as come up with robust boolean algorithm for the nurbs brep solids.

Thanks!

iamnotstone, 2017-09-08:
OK, thanks for your comments. I'll try to follow your advices, and try them out.I'll let you know if it works. 
And I want to join in your next nurbs-based project. I am once a mechanical engineer and now changed to a cad/cae soft engineer. I have one year of coding experience based on opencascad , freecad and threejs .And I am familiar with c++, javascript, python and mongodb. I can be a good assistant for you.

mmiscool, 2023-03-19:
Issue is very old. Overcome by events.

closed 2023-03-19 by mmiscool