xibyte/jsketcher #53: Implement boolean algorithm NURBS based BREP

xibyte, 2017-09-08:
The current BREP boolean algorithm works only for polyhedrons. To move forward we need to support NURBS surfaces and curves.

xibyte, 2017-09-08:
Also current algorithm isn't very robust and fails in some corner cases.

iamnotstone, 2017-09-08:
After think twice about the whole things.I think it's a timing to start a new project which  include brep, csg algorithms and triangulation algorithms. And take advantage of algorithms supported by verb-nurbs which include tessellate and intersection algorithm of nurbs surface.

I have to learn more about nurbs-surface/curve before doing this.

By the way, triangulation of plane face with nurbs-curve in outer loop can easily improved by add the code below in '.../3d/triangulation.js'

export function TriangulateFace(face) {                                                     
  function arr(v) {                                                                         
    return [v.x, v.y, v.z];                                                                 
  }                                                                                         
                                                                                            
  function vertexCallback(data, out) {                                                      
    out.push(data);                                                                         
  }                                                                                         
  function combinecallback(coords, data, weight) {                                          
    //throw 'should never happen cuz brep is non-manifold'                                  
  }                                                                                         
  function edgeCallback(flag) {                                                             
  }                                                                                         
                                                                                            
  var tessy = new libtess.GluTesselator();                                                  
  // tessy.gluTessProperty(libtess.gluEnum.GLU_TESS_WINDING_RULE, libtess.windingRule.GLU_TESS_WINDING_POSITIVE);
  tessy.gluTessCallback(libtess.gluEnum.GLU_TESS_VERTEX_DATA, vertexCallback);              
  tessy.gluTessCallback(libtess.gluEnum.GLU_TESS_BEGIN, begincallback);
  tessy.gluTessCallback(libtess.gluEnum.GLU_TESS_ERROR, errorcallback);                     
  tessy.gluTessCallback(libtess.gluEnum.GLU_TESS_COMBINE, combinecallback);                 
  tessy.gluTessCallback(libtess.gluEnum.GLU_TESS_EDGE_FLAG, edgeCallback);
  
  const normal = arr(face.surface.normal);                                                  
  tessy.gluTessNormal(normal[0], normal[1], normal[2]);                                     
  
  const vertices = [];                                                                      
  tessy.gluTessBeginPolygon(vertices);                                                      
  
  for (let loop of face.loops) {                                                            
    tessy.gluTessBeginContour();                                                            
    for (let e of loop.halfEdges) {                                                         
+      if(e.edge.curve.isLine)
        tessy.gluTessVertex(arr(e.vertexA.point), e.vertexA);                               
+      else if(e.edge.curve.isNurbsCurve){       
+       tessy.gluTessVertex(arr(e.vertexA.point), e.vertexA);                               
+        let nurbsVectors = e.edge.curve.verb.tessellate()
+       nurbsVectors.shift()

+        nurbsVectors.forEach(v => {
+           let midVertex = new Vertex(v[0], v[1], v[2])
+           tessy.gluTessVertex(v, midVertex)}
+        )
      }
    }
    tessy.gluTessEndContour();
  }
....

xibyte, 2017-09-08:
I don't think it will work. The tessellation algorithm is fundamentally 2D algorithm and it works/deals with contours on a PLANE. What's why it's supposed to be initialized with a normal 
const normal = arr(face.surface.normal);
tessy.gluTessNormal(normal[0], normal[1], normal[2]);

This is the part for the gluTess documentation 
_The gluTessNormal function describes a normal for a polygon that you define. **All input data is projected** onto a plane perpendicular to one of the three coordinate axes before tessellation, and all output triangles are oriented counterclockwise with respect to the normal._

It's not even possible to tessellate a nurbs without any boundaries using the libtess.

As I described in the other issue ticket the tessellation of bounded NURBS is only possible in parametric domain with the curvature analysis of the triangles after. See my comment here https://github.com/xibyte/jsketcher/issues/51

eric-schleicher, 2017-09-08:
see [verbnurbs](http://verbnurbs.com/) by @pboyer

iamnotstone, 2017-09-09:
@xibyte Thanks for your explain on gluTessNormal. The code before is just for a plane face with nurbs-curve in its ourerLoop. I know it do not works for nurbs -surface. In fact, I found tessellate algorithm was supported in verbnurbs by @pboyer .So I think if we can adjust the structure of our code and make use of the algorithm in verbnurb directly ? May be we can even adjust the code in the verbnurbs?

xibyte, 2018-01-03:
done in 8204748ad32e35176af5dfff46f30d351c87e619

closed 2018-01-03 by xibyte