xibyte/jsketcher #59: Feature Request: Trim sketch curves

mmiscool, 2017-11-13:
Tool in the sketcher to allow for trimming of sketch geometry and creation of constraints.

Selecting a segment would remove that segment until it intersects with another segment or that segment ends. If the segment is trimmed to an existing segment what ever remains shall have a constraint created for either a point on curve or point on point. 

I can provide a video demonstrating the desired affect. 

Would also like a split segment tool that could break a line or curve in to multiple lines and curves based on selection of cutting curve and curve to be cut. No geometry would be destroyed. Just broken in to individual segments.

ezequielleonzybert, 2024-05-09:
Hi, is this now implemented?

mmiscool, 2024-05-10:
Not currently.