xibyte/jsketcher #64: Trying to port a lot of Brep in Java.

MrPyne, 2018-03-23:
Hi. Possible not an issue but I can't find away to contact any of the developers for this project.

 Am trying to write a Brep package in Java and I was using jsketcher as a base line.  Anyways, am having some issues and am not sure if am miss understanding or something is broken. I have pretty much everything in Brep written in Java and am now able to test it out but it's not working. Question is the current Master branch even in a working condition or is under construction? If Master is functional than my issue is in BooleanOperations.intersectFaces(). Seems like this is not working. When it goes to find the Nodes for the CurveEdgeIntersection it fails to find some even though there was a Surface to Surface intersection. Either the Nodes vertex was all ready made in a previous intersection or the splitting fails. Splitting seems to fail if leaves occurs before enters.

Am testing with a thin tall rectangle in the middle of smaller wider rectangle and subtracting. So the result should be the small wider rectangle with a rectangular hole through it.

Any help would be welcome. If this isn't the place for this type of thing let me know. Maybe we can set something else up. I will try and post my Java code to gitlab if needed. Thanks.

MrPyne, 2018-03-25:
So i think have most of it working now. I noticed verb for java might not behaving right. There looks to be numerical issues. Most concerning is when I do the surface intersection of two flat(nurb surfaces) I get 2 curves. Both are lines but i would of expected just one curve.

closed 2018-03-25 by MrPyne

MrPyne, 2018-03-25:
First off thanks for replying. 

Yeah I see what your saying about verbs intersection. At one point it was trying to make cubic interpolation out of 2 points. Anyways, I did confirm am having more troubles than what you described. My big thing right now is verb compiled to Java is not working as it does for JavaScript. The same two surface's intersections are different between Java and JavaScript. I don't think verb can be compiled to Java with out a lot of manual effort to fix some of these issues. 

Am playing around with the idea of rewriting what I have done in Haxe. Using your work done in Jsketcher to basically recreate a Brep package in haxe using verb. My visualization would most likely be done with Openfl by the looks of it. At this point maybe I could do something directly to verb to fix some of those issues with surface intersection. But maybe I just do what you explained above. Who knows.

I'll keep an eye on this project. Thanks for your work too.

xibyte, 2018-03-25:
Thank you. If you like to chat about it
https://www.facebook.com/valentin.yerastov

<https://github.com/xibyte/jsketcher/issues/64#issuecomment-375993709>, or mute
the thread