xibyte/jsketcher #67: Import DXF file from local

StevenKung, 2018-05-24:
Hi xibyte

This app is awesome!
I have a question about how can i import the DXF file from local?
thanks

xibyte, 2018-06-16:
There is no way currently. However it is a nice feature to have and it's definitely on my list of things to implement. Thanks for the suggestion I can start working on it as soon as I am done with the 3d brep engine.

TheMBeat, 2018-08-03:
Hey xibyte and mmiscool,

how about the feature? 

thanks

mmiscool, 2020-03-25:
On the radar but not at top of priority list.

mmiscool, 2022-12-05:
Basic DXF import functionality implemented. 
[image]

closed 2022-12-05 by mmiscool

jtcheng, 2023-01-12:
@mmiscool 
There're two issues for import with dxf file:
1. the rectangle elements are loss.
2. no constraint is created (detect) from the import dxf file, we need to manually create all the constraint, this is not easy for large dxf.

mmiscool, 2023-01-18:
@jtcheng Try the latest dev branch. It should be working now.