xibyte/jsketcher #73: web-cad.org version

kennethshawfriedman, 2018-10-22:
I see the current version of the repo in the master branch is quite a bit different from the one on web-cad.org. Is there a way to access the exact code that's live on the website?

xibyte, 2018-10-22:
Hi Kenneth,

I am doing active development in the master branch so things may become
unstable. The commit web-cad is build off of is
53f3e8d06c52c521a75c570c544c4a69939d79d7
if you want to play around with it.

I am actively working on new version which is going to be out in a month.
It will have a new BREP engine and will have more functionality.
I haven't published the new engine yet so master branch is quite a bit
inconsistent but I am working on it.

Thanks,
Val

mmiscool, 2020-03-30:
See #96 for when latest version will be provided.  
Closing old request here as the link to the version used for the site at that time was provided. 
https://github.com/xibyte/jsketcher/commit/53f3e8d06c52c521a75c570c544c4a69939d79d7

closed 2020-03-30 by mmiscool