xibyte/jsketcher #74: Feature Request:

Arredatore, 2018-11-15:
Is it possible to make exposing parameters like here [Allegorithmic exposed parameters](https://support.allegorithmic.com/documentation/sddoc/expose-a-parameter-102400062.html)

And then use the editor in a "viewer mode" with only exposed parameters visible? Something like this [https://codepen.io/ShapeDiver/pen/WRZQRo](https://codepen.io/ShapeDiver/pen/WRZQRo)

That will be great for the online shops like these [https://junddo.com/meuble-vinyle-sur-mesure-portee/](https://junddo.com/meuble-vinyle-sur-mesure-portee/) [https://medo-arauco.com/en/collections](https://medo-arauco.com/en/collections) [https://www.mipuf.es/palets-pufs-p-396.html](https://www.mipuf.es/palets-pufs-p-396.html)

mmiscool, 2020-03-25:
Unification of an expression engine between 2d sketcher and 3d feature parameters is underway.

The expression editor UI is present and works for 3d feature parameters but not 2d sketches as of today.

A customizable method of exposing specific expressions in a UI that is custom to a template part is not going to happen untill the expression engines between the 3d and 2d sides are fully unified.  This kind of feature is similar to some OpenSCAD part configurators or Siemens NX product template studio. 

This is also related to part family geometry generation.

mmiscool, 2020-03-25:
See https://github.com/xibyte/jsketcher/issues/39 for info related to expressions progress.