xibyte/jsketcher #79: Does STL Import work for anyone?

onsombal, 2018-12-04:
Has anyone gotten STL Import to work?

STL Export works fine for me.

Evil-Spirit, 2018-12-17:
@mmiscool

Evil-Spirit, 2018-12-17:
@mmiscool Let me introduce https://github.com/NoteCAD/NoteCAD

zeser78, 2019-01-16:
No, Do you find a way to import?

onsombal, 2019-01-16:
No, nothing new since @mmiscool replied that it's not working. I'm assuming it must be on the road map. It would be such a good feature!

mmiscool, 2020-03-25:
Right now BREP and step file import/export is higher on the priority list. With much of the current work being done and the switch to using BREP rather than meshes for booleans. Using geometry from STL files will require some work where BREP is used for any new geometry and modifications and a secondary process to mesh the BREP solids and perform mesh booleans on the original STL. It is on the radar but not a priority at the moment.