xibyte/jsketcher #82: Licence usage Issue

Roba1993, 2019-03-25:
Hi xibyte,

this library/application is really good and I would like to use it for another software.

The goal is to use your library (without any changes) and leverage it within another application.
With the actual licence, I would need to provide the code/use same licence for the other software as well.

Are you open for an Apache or MIT licence, or any licence which allows the use without changes and don't forces to provide the code for the application around it?

Thanks,
Robert

dumblob, 2019-11-28:
I'm in a similar situation. Would you @xibyte be willing to switch to MIT/BSD?

mmiscool, 2020-03-24:
The licence is intentional. If some one or an organisation would like to licence it with out meeting the requirements of the  APLG 3.0 a commercial licence would have to be used. At this point a commercial licence is not available but if there is interest it might be possible to arrange. 

The engine is still under development and an assembly module is in the pipeline.

Roba1993, 2020-04-01:
Okay thanks for clarification.

closed 2020-04-01 by Roba1993

mmiscool, 2020-07-11:
Hello,
We have started working on packaging to make distributing and using the core cad technologies behind JSketcher available to be used in commercial environments. 

See https://www.autodrop3d.com/parametric-cad.html