xibyte/jsketcher #91: Change the line color, is it possible?

webblufter, 2020-01-01:
Is is possible to change the line (lines) or drawing color instead of keeping it default as white?
What if I want to highlight things/lines or walls with different colors?

vikm2o, 2022-12-27:
@mmiscool any updates on the enhancement ?

mmiscool, 2022-12-27:
Storing attributes against faces, edges and sketch segments is planned for. 

After that is implemented it will be possible to track and store information such as color along with other metadata related to thread call outs, capture of symbolic information for GD&T visualization and driving tool/process selection in automated CNC path creation.

These attributes also have applications to procedural generation of architectural elements like walls. 

Members of the open-collective help prioritize development using the projects page.
https://opencollective.com/jsketcher-ad3d/projects