xibyte/jsketcher #93: sample files

ThorsenRune, 2020-02-21:
Hi, this project looks great. May I propose to put some (five) sample models that can be loaded for visitors to play with. ( Im looking for an alternative to Onshape for open codesing projects)

mmiscool, 2020-03-28:
@ThorsenRune  Do you possibly have some particular mechanical drafting examples that you would like to see? Preferably old text book exercises that might be in the public domain.

Examples like that would assist in making test cases for the automated test frame work.

ThorsenRune, 2020-03-30:
As I think this project could be great for solving the co-design issue of affordable 3D design I tried it out. Unfortunately I was stuck in not knowing how to edit a sketch as the workflow seems different from what I know. I was looking for help allover but did not find any. 
As for examples  can I send you a private message and how?

mmiscool, 2020-03-31:
The current version has 3d sketching broken at the moment. You can get in to the fallback 2d sketcher.

The normal process to start building a part is to add a plane the the model. Select the plane and click the sketch icon.

In the top right hand corner of the graphics area you should see a tool bar with 3 buttons. If you click the center button it will open up the sketcher tool in 2d mode. 

Once you are done with the sketch in the 2d mode you should click the save button and exit out of the 2d sketch tab located at the bottom of the screen. I will make a little graphic to show this. 
[image]

ThorsenRune, 2020-03-31:
Hi mmiscool, thanks for your answer. I havent got so far learning the program yet so I dont know what you intend by 3d sketching. Great with the animation. Unfortunately Im on a laptop so its hard to see and I havent even got to the extrusion point yet. 
I have made a 1 page document about my initial experience if it can be helpful to understand users expectations. https://docs.google.com/document/d/1PGW2oKL6YULiER7IGxOojSPa7gDWvxKFMCluyaJr_bg/edit

closed 2020-03-31 by ThorsenRune

mmiscool, 2020-03-31:
looks like you have permissions turned on for that doc