xibyte/jsketcher #99: Constraints stop working.

z26, 2020-05-01:
Hi,

I don't know how this happens exactly, but at some point while editing a sketch in the dedicated sketch view, the constraints stopped working (as in they still exist, but they are ignored when moving sketch points around with the mouse)

I'll edit this post with more info if I replicate this issue, intentionally or accidentally.

xibyte, 2020-05-03:
Are you using a build from the latest master branch? If so it is currently
broken in the master branch and I am fixing it.

z26, 2020-05-04:
I'm going to the the web-cad.org website by typing the url.

mmiscool, 2020-05-07:
The web-cad.org is periodically updated. There is a bunch of UI stuff and bug fixes happening now so it will be updated after the master branch is in a good state.

mmiscool, 2022-11-13:
@z26 
This is a very old issue. Could you confirm this is still a problem or not? Seeing if we can close this one.

z26, 2022-11-13:
It's been quite long so I don't remember the exact steps I took back then, but i am unable to recreate this issue.

While investigating this, I did find that setting a distance constraint to A then writing something wrong on purpose like "A = x" (if x doesnt exist) will cause the app to become somewhat but not fully unresponsive. Not a serious problem (the situation is recoverable by deleting the offending statement) but might as well mention it regardless.

mmiscool, 2022-11-14:
Ok. Closing this one for now. If you are able to replicate go ahead and create a new issue.

closed 2022-11-14 by mmiscool