xibyte/jsketcher #106: Dimension lines problems

itsanov, 2020-06-02:
Dimension lines do not zoom together with model.
Dimension scale not working.
If dimension is selected and backspace is pressed while changing "Dimension scale" - dimension line is deleted.
Dimension lines are not exported together with model.

mmiscool, 2020-06-03:
Are you running locally from source or using web-cad.org?

itsanov, 2020-06-03:
Locally!
Cloned the "master" brunch of repository.
npm install
npm start

mmiscool, 2022-10-01:
Is this still an issue for you? 
If so could you record a screen video showing the problem and post here? Not sure I understand what you are referring to with the dimension scale.

mmiscool, 2022-11-08:
Closing old issue. Additional information not provided or issue is no longer present.

closed 2022-11-08 by mmiscool