xibyte/jsketcher #108: The Big Hold Back

wanfuse123, 2020-06-20:
Hi! I would like to say your work is incredible!

I have long thought it would be nice to be able to build this sort of thing from a browser or server from a local web server.

Unfortunately, this is not my area of expertise.

I do have one thought/suggestion.

All of the open source programs have one very fatal flaw. Mainly the limited support of input and output file formats. 

I am curious, what would it take to support formats for major software vendors such as autocad and solidworks and rhino and for formats such as DXF and 3D DXF (If there is such a thing)?

Is it a question of purchasing rights to develop for these formats or is it a physical resource limitation (as in not enough programmers?) or more of a monetary limitation?

I am curious as to what it would take to make this happen? 

I think it would be great if there was more community donations and financial support for such endeavors. Perhaps starting a gofundme site or or something?

Regards,

Steven

xibyte, 2020-06-22:
Hi Steven,

Thanks a lot for sharing your ideas. I am currently working on external
repositories support and after getting it done it'll be
theoretically possible to run the application locally against a local web
server and having models served through that server.

As far as importing goes I just implemented the STEP format which is an
industry standard for exchanging 3d models and it'll cover most of the
needs of  importing(adn exporting too) of external models from/to
Solidworks or Rhino or etc. It all will be part of the next release. So
yes, that's definitely the direction we are going.

Val

mmiscool, 2022-08-27:
Duplicate of #67 

Closing this issue. 

See https://github.com/xibyte/jsketcher/issues/67 for when this will be resolved.

closed 2022-08-27 by mmiscool

mmiscool, 2022-12-06:
See https://github.com/xibyte/jsketcher/issues/67 for first cut on DXF import functionality and screen shot showing how to import in to the 2D sketcher.