xibyte/jsketcher #109: Does  support direct opening of dwg documents?

yoyomule, 2020-06-22:
Hello, does this project support direct opening of dwg documents?

mmiscool, 2020-06-23:
Not at the current time but STEP file input is coming soon.

yoyomule, 2020-07-03:
Thanks.I'm thinking about whether I can use it as an online browsing tool for cad documents, not just an editor, such as opening dwg. If not, can dwg browsing be implemented by supporting svg? (I can convert dwg to svg first)

mmiscool, 2020-07-12:
Problem same as https://github.com/xibyte/jsketcher/issues/108

Any cad program or viewer will have to work with in the confines of the limitations of the hardware (ram, storage, CPU). To this point large documents will have to work in these limits unless processing is offloaded to a server.

mmiscool, 2022-12-07:
See https://github.com/xibyte/jsketcher/issues/67 for first cut on DXF import functionality and screen shot showing how to import into the 2D sketcher.