xibyte/jsketcher #111: can't figure out how to revolve

dominictarr, 2020-07-02:
I managed to use extrude and loft, but I can't figure out how to make revolve work. I tried to make a torus. I created a plane and drew a circle on it. I could see that I needed to select two things. I clicked on my circle and it seemed to take it, but then for axis I tried to click a datum pointer that was parallel to the face of the circle... but it wouldn't fill the "axis" slot.

then I tried drawing another circle, on a perpendicular plane, this did select as the "axis" but nothing happened.

I don't know what I'm doing wrong, the software looks really good! I can't figure out how it works.

mmiscool, 2020-07-12:
In the sketch you must create a construction line to use as the center line of the revolve. 

You can select a regular line or a construction line to be the axis 
[image]

mmiscool, 2020-07-12:
Example making a sketch with a construction line in it. 
[image]

mmiscool, 2021-06-13:
No further activity on this issue and demonstrations were provided. Closing.

closed 2021-06-13 by mmiscool