xibyte/jsketcher #112: cannot subtract torus from aligned cylinder

dominictarr, 2020-07-02:
I am trying to create a cylinder with a groove in the outside. I think I have found a bug.
If the torus is around the cylinder, and I try to subtract the torus from the cylinder, it doesn't subtract anything.
but if the torus is off center by even 0.1 then it does work.

here is the downloaded json (github wouldn't allow me to attach it)
```
{
"model":
{"history":[{"type":"TORUS","params":{"radius":250,"tube":50}},{"type":"CYLINDER","params":{"radius":100,"height":250}},{"type":"SUBTRACT","params":{"operandA":"S:1","operandB":"S:0"}},{"type":"DATUM_CREATE","params":{"x":"0","y":"0","z":"0"}},{"type":"DATUM_MOVE","params":{"datum":"D:0","x":"0","y":"30","z":"0.1","copy":false}},{"type":"DATUM_ROTATE","params":{"datum":"D:1","axis":"X","angle":"90"}},{"type":"TORUS","params":{"datum":"D:2","radius":"100","tube":"20"}},{"type":"SUBTRACT","params":{"operandA":"S:2","operandB":"S:3"}}],"expressions":""},
 "sketches": []
}
```

I couldn't see if there was a way to import a file, but I did manage to set this to `localStorage` (by taking the `model` property) and then open it in a new tab.

It looks like there is an extra datum in there? is it that the datum rotate is always  a copy? I was getting some "datum does not exist" problems when messing around trying to get this to work.

mmiscool, 2020-07-12:
If you are using the latest dev version the download option is available.

I have verified that there is some sort of bug relating to this. Marking as bug. 
[image]

mmiscool, 2022-03-29:
No longer an issue in MDF branch.

closed 2022-03-29 by mmiscool