xibyte/jsketcher #116: Import step or brep file

ancorasir, 2020-07-23:
Is it possible to import step file from local computer?

I used the "Import step file" icon but fail to import any step file. Could please give an example? Thanks!
@xibyte

mmiscool, 2020-07-24:
I will have a video up soon about how to do this. Stay tuned

xibyte, 2020-07-24:
Currently you can only specify an URL to a file. So your step file should be served from some HTTP server. If you are running the jsketcher as a dev server you can just drop your step file inside web directory and then specify http url to that file in the wizard. Hope it helps.

ancorasir, 2020-07-26:
I’m able to open some step files I uploaded on GitHub. However, I found that jsketcher fail to read some other step files for example 

https://raw.githubusercontent.com/ancorasir/node-occ/master/7.STEP

I can read the above file with another occ repo so the file so I’m wondering if the problem is with webassembly building of occ. Any suggestions? Thanks.

mmiscool, 2021-06-13:
Import of step file from local file system now possible. 
Imported step files will now show as a feature in the part history. 
Boolean and edge operations are able to be used with the imported object.

0xCoto, 2022-04-08:
> Import of step file from local file system now possible.

@mmiscool Could you please elaborate how this is possible on the demo page (http://web-cad.org/)? `File` -> `Import Project` -> [my step file] doesn't seem to work.

mmiscool, 2022-04-16:
@0xCoto As it stands today the JSketcher build deployed in the demo is quite outdated. We are working on the next major release currently but there is no ETA on the next deployment to the demo site.

mmiscool, 2022-08-06:
This is a link to a recent build of the dev branch. It includes step, brp and FCStd file import functionality.

For now be careful not to import any thing with extremely large numbers of edges as that might result in a laggy viewport. There is a performance enhancement in the works to address this as we are migrating the very old version of threejs to the latest and greatest.

mmiscool, 2022-08-06:
Woops. Forgot to past the link in the previous comment. 
https://autodrop.cloud/jsketcher/

closed 2022-11-02 by mmiscool