xibyte/jsketcher #127: Can you provide the import function of SolidWorks model? I can pay for it

WREQI, 2021-03-02:
Can you provide the import function of SolidWorks model? I can pay for it

mmiscool, 2021-03-03:
Hello,
We do have import capability from STEP file. All major cad software packages support step file export as this is the preferred interchange format. Importing directly from native solids works would require a large amount of effort and replicating all of the features that solid edge supports to properly import and interpreter the part history. 
Would be happy to talk and discuss your use case and see if we can help but a fully blown solid works import capability is not within the capabilities.

WREQI, 2021-03-09:
How can I contact you quickly

WREQI, 2021-03-09:
Discord has expired

WREQI, 2021-03-09:
I can't reach you on the website

WREQI, 2021-03-09:
Can you give me your phone number

mmiscool, 2021-03-11:
I replied to your email. You might have to check your spam folder.

closed 2021-04-07 by WREQI