xibyte/jsketcher #138: Is it impossible to import a sketch from SVG/DWG etc?

nikkwong, 2021-06-11:
I have an external drawing that I'd like to bring into this program. Is that possible? Also curious if the project is alive, it looks great! Thanks

mmiscool, 2021-06-13:
Hello, 
Project is alive and looking for funding to be able to accelerate development. 

DXF and SVG are not currently available as import formats however step file support was recently added. 
I am marking this as an enhancement.

mmiscool, 2021-06-13:
See this issue also. Seems to be duplicate. 
https://github.com/xibyte/jsketcher/issues/108
https://github.com/xibyte/jsketcher/issues/109

closed 2021-06-13 by mmiscool

mmiscool, 2022-12-06:
See https://github.com/xibyte/jsketcher/issues/67 for first cut on DXF import functionality and screen shot showing how to import in to the 2D sketcher.