xibyte/jsketcher #160: FCStd?

samuk, 2022-05-17:
Unclear if this is a dead project?

The idea of a [WebUI for FreeCAD came up here](https://forum.freecad.org/viewtopic.php?p=595443&sid=0cbfaf3d2ae44ce2bb725d71a39618f2#p595443) would doing that work be of interest to current or new developers ?

mmiscool, 2022-05-17:
Hello. I sent you a private message on the free cad forums.

closed 2022-05-21 by samuk

mmiscool, 2022-07-29:
This feature has now been added.
[image]

samuk, 2022-07-30:
Nice. Will check it out!

samuk, 2022-07-30:
Maybe https://github.com/yorikvanhavre/WebTools too?

mmiscool, 2022-08-23:
@samuk Here is a video showing how to import from FCStd 
https://youtu.be/VjoP9uSRSYs

It should work in the recently updated build running here.
https://autodrop.cloud/jsketcher/