xibyte/jsketcher #171: Deployed live demo is very out of date

MattFerraro, 2022-08-06:
The version of JSketcher that is currently deployed at https://web-cad.org/ is very, very old. It would be much easier for new users if that version were kept up to date with master.

Also, it seems like most of the work is going on in the 'dev' branch. Maybe there could be a second URL which keeps a hosted version of the dev branch?

mmiscool, 2022-08-09:
A much more recent build can be used here.
https://autodrop.cloud/jsketcher/

Update to web-cad.org will happen eventually as we retool the automated build system.

samuk, 2022-08-19:
Does the autodrop instance support FCStd import? If so how should I use it please? https://github.com/xibyte/jsketcher/issues/160#issuecomment-1198938774

mmiscool, 2022-10-01:
The https://web-cad.org demo has been updated.

samuk, 2022-10-01:
With more complex parts this causes Firefox to 100% CPU and then often crashes. Is there some optimisation possible?

Example file: https://github.com/rosmo-robot/Rosmo_3D/blob/main/V4/camcnewParedDownH.FCStd

mmiscool, 2022-10-11:
Currently the feature history is executed in the main thread on the browser. 
We are currently working on breaking out the execution of the feature history to run in a non blocking web worker. Also to note it looks like you are importing an assembly and I think it's really great that it is loading and visualizing. We don't currently have the JSketcher assembly module working yet. When we start working on doing assemblies there will be some significant changes as parts will have to be generated for each individual object.

[image]

samuk, 2022-10-18:
Is it possible to have shareable a link that would load an assembly?

mmiscool, 2023-12-16:
https://maker.cloud/jsketcher/

The build hosted there is based on the dev branch.