xibyte/jsketcher #176: is all source code open source

alexbrownsoft, 2022-08-08:
I am a bit confused about the the open source status of this software after reviewing the commercial license.

Without paying commercial license, is the source code of this software including the wasm file avaiable as open source?

Or only after the commercial license is paid can we get all the source codes?

mmiscool, 2022-08-09:
Full source code is on github release under the licence linked here.
https://github.com/xibyte/jsketcher/blob/main/LICENSE

You may use the software as is without any limitations. 
If you modify the software the only requirement is that you share back your changes so that they might improve the main project here.

Commercial licences are available if you wish to build a proprietary close source application with out the requirement to contribute code changes back to the project. 

Do you have a particular situation or concern? Happy to chat on discord or set up a google meet.

mmiscool, 2022-08-11:
Closing issue. Please go ahead and reopen if you still have questions.

closed 2022-08-11 by mmiscool