xibyte/jsketcher #177: The exported dxf not opening in AutoCAD

tarikjabiri, 2022-08-18:
Hi,

I looked into the `dxfExport`function and I found this [line](https://github.com/xibyte/jsketcher/blob/main/web/app/sketcher/io.ts#L457) should be deleted because the function `generateAutocadExtras()` already called in the [Drawing.constructor()](https://github.com/ognjen-petrovic/js-dxf/blob/master/src/Drawing.js#L50). I think this is the main reason, I didn't test it.

I want to make a suggestion: I see you have [`Dimensions`](https://github.com/xibyte/jsketcher/blob/main/web/app/sketcher/io.ts#L452) also needed to be exported to the dxf. the `dxf-writer` does not support dimensions. Currently I am working on an implemention of dimensions but in my lib [dxfjs/writer](https://github.com/dxfjs/writer), at this moment I didn't push the updates to github but this will be soon. What I am saying is if its ok to switch to another lib I will help you making the dxf export work and also support all geometries. If not I will make PR to make the dxf generated from `dxf-writer` work in AutoCAD if the problem not fixed by removing this [line](https://github.com/xibyte/jsketcher/blob/main/web/app/sketcher/io.ts#L457).

Regards

mmiscool, 2022-08-20:
Thanks. It looks like there was no problem swapping over to your implementation. 

It looks like you have been working on dfx parsing library. I imagine that it might be possible to leverage that to create an import tool for JSketcher. What is the current status on that? Is it at a point where it might be used to extract 2D geometry from a standard dxf?

tarikjabiri, 2022-08-20:
Yeah I am working on it. But its not ready for now, will be in the future.
I need the parser also to be able to read the existing files and update them in the dxf writer.

mmiscool, 2022-08-27:
The pull request has been merged. Closing this issue.

closed 2022-08-27 by mmiscool