xibyte/jsketcher #179: Export as FCStd

samuk, 2022-08-19:
Would it be possible to support export in FCStd format as well as the import?

https://github.com/xibyte/jsketcher/issues/160#issuecomment-1198938774

mmiscool, 2022-08-19:
We can do an export to brp. That is the OpenCASCADE internal format. FreeCAD opens those files with out any resistance. A FCStd file is just a zip file containing some XML and a collection of brp files. 

Would brp export fit your use case or do you absolutely need FCStd output?

samuk, 2022-08-29:
Yes it sounds like brp would do the job. Thanks

mmiscool, 2022-08-29:
Ok. I think we can get this working relatively soon. Stay tuned for when this will drop.

samuk, 2022-10-12:
Actually thinking about this, proper FCStd would be preferable. I'm thinking of future integration with Github/Gitlab where you could load your FCStd from a git URL and then save your changes to Git.

Does that seem at all viable/on your roadmap?

mmiscool, 2022-10-14:
Import / export functionality relating directly to FCStd is really only for single direction data migration. It would not be especially good for active community design as each time a model is loaded or stored only the geometry is preserved. The part history would not be preserved going from one CAD application to another.

samuk, 2022-10-15:
Is it possible to record the part history in a FCStd? I realise it's probably low-priority work. I do think frictionless compatibility with FreeCAD would aid adoption and resolve some use cases where FreeCAD currently has more functionality.

Put another way: If Jsketcher fulfils 90% of my design needs I'm likely to use it, and do finishing work in Freecad.

If I can't do finishing work in Freecad, then I'd probably have to use it for the whold project from start to finish.

mmiscool, 2022-10-18:
Well for now I pushed the "Export BREP" command. 
If you run from the dev branch you can give it a try.

mmiscool, 2023-03-13:
@samuk Was the export brep adequate for loading models produced in JSketcher to FreeCAD?

samuk, 2023-03-14:
To be completely honest I found working with FLOSS cad tools so frustrating, I've just started working in ways that don't require 3D modeling. I'll try again in a couple of years, but the learning curve was too steep.

mmiscool, 2023-03-15:
Closing issue for now.

closed 2023-03-15 by mmiscool