xibyte/jsketcher #186: Is the the open source part of the license more like GPL or LGPL?

jirihon, 2022-10-03:
Hi, thank you for all the work on jsketcher, especially on the parametric 2D sketch engine that is absolutely unique. I think there is no open-source project that would be so full-featured like this while being implemented in JavaScript.

But here is my question. Could you please clarify the application of jsketcher license in terms of using parts of the code as a library in the context of a larger project? I see there is the statement `Any modifications made to the software must be submitted to Autodrop3d LLC`, but what does it mean exactly? Can I isolate the jsketcher code (for example the 2D constraint solver) into an open source library and use it freely in any project with different license? Or the whole project that will be using the jsketcher code should be public, pushed as PR to the jsketcher repository, and sharing its license? Or in other words, is jsketcher license viral like the GPL or more permissive for inclusion in other projects like LGPL?

mmiscool, 2022-10-06:
Currently the licence is more similar to GPL in that any derivative project must be released under the same licence. We do however provide a clean path for commercial usage and integration. 

The reasoning behind this decision is driven by the desire to secure funding for further development and ensure businesses looking to commercialize in a closed source system fairly support the work. 

We would be interested in having a more in depth discussion on the possibility of a bit more flexibility for a library use case. If you would like to set up a call to discuss please reach out using this form. 
https://www.autodrop3d.com/support.html

mmiscool, 2022-10-21:
If you have additional questions please reopen this issue or reach out.

closed 2022-10-21 by mmiscool