xibyte/jsketcher #194: Docker and improvement

julio-matarranz, 2022-12-21:
Dear,

I made a Dockerfile to build "dev" branch (the git clone could be replaced with a COPY).
It's not the best because of webpack that doesn't package everything in the dist folder... but I managed to get a working jsketcher on a nginx with that :

```
FROM node:latest as build

WORKDIR /usr/src/app

RUN git clone --branch dev https://github.com/xibyte/jsketcher.git .

RUN npm install 
RUN node ./node_modules/webpack/bin/webpack.js --config webpack.config.js --progress --profile

FROM nginx:stable-alpine
COPY --from=build /usr/src/app/dist /usr/share/nginx/html
COPY --from=build /usr/src/app/web /usr/share/nginx/html
COPY --from=build /usr/src/app/node_modules/jsketcher-occ-engine /usr/share/nginx/html/lib-assets/jsketcher-occ-engine
COPY --from=build /usr/src/app/node_modules/font-awesome /usr/share/nginx/html/lib-assets/font-awesome

EXPOSE 80
```

Could you please improve the webpack configuration to make a better packaging in the "dist" folder ?

Also, just to share, my simple docker-compose :

```
version: "3"
services:
  jsketcher:
    image: jsketcher:latest
    restart: unless-stopped
    container_name: jsketcher
    hostname: jsketcher
    ports:
      - "80:80"
```

mmiscool, 2022-12-22:
Thanks for pointing out the web pack configuration issue and the libraries that fail to be copied over to the dist folder. We will be looking in to this and try to get it all working correctly.

mmiscool, 2023-03-13:
Latest dev branch is working for running 'npm run build' producing the dist folder that can be copied to a static CDN host.

closed 2023-03-13 by mmiscool