xibyte/jsketcher #197: Chrome and Safari performance is much worse than Firefox in 2D sketcher

conradlempert, 2023-02-06:
In the 2D sketcher with many objects and constraints in the scene.

Firefox runs smoothly, Chrome and Safari give me about 2 FPS.

I think this is worth investigating since Firefox shows that much better performance is possible.

Chrome 109
Safari 15
Firefox 109
macOS 12.0.1
Mac Mini M1

mmiscool, 2023-02-06:
@conradlempert Thanks for reporting this issue. 

Is there any chance you could could capture a short screen recording video of the behavior?

Also could you export the JSON for the problematic sketch and attaach it to this issue so that it can be investigated? 

Can you indicate if you are using the main branch, dev branch or the https://web-cad.org demo? The web-cad.org demo is a bit behind.

mmiscool, 2023-02-13:
@conradlempert Just following up on this. 
Can you provided any more information about this and any sketch information data export so that we can test in chrome. I run a chromebook my self and use chrome for testing. I typically don't use firefox for testing.

closed 2023-03-05 by mmiscool