xibyte/jsketcher #198: Jsketcher sketch geometry doesn't visualize after finishing sketch

Cooper6, 2023-02-27:
name: Elec box lid v2
description: Jsketcher sketch geometry doesn't visualize after finishing sketch

[image]
[image]
 
Jsketcher .json file downloaded cannot be uploaded to Github.com.

Cooper6, 2023-02-27:
I changed the .json to .txt.  Please see attached.

[Elec box lid v2_63fbe7f79165120021deeee0.txt](https://github.com/xibyte/jsketcher/files/10842808/Elec.box.lid.v2_63fbe7f79165120021deeee0.txt)

mmiscool, 2023-03-05:
Issue seems to be the same as this previously reported issue.
https://github.com/xibyte/jsketcher/issues/195 

Thanks for providing the additional test data.

mmiscool, 2023-03-12:
[image]
Fixed in dev branch. 
Will be updated on autodrop.cloud some time in the next 2 weeks.

closed 2023-03-12 by mmiscool

Cooper6, 2023-03-13:
Thanks Mike.

CC