xibyte/jsketcher #199: performing operation with current parameters leads to an invalid object (self-intersecting / zero-thickness / complete degeneration or unsupported cases)

oguzhannkayaa, 2023-03-02:
name:error
during import the .dwg format image.

mmiscool, 2023-03-05:
Hello,
Import of .dxf files works for basic entities like line work, arcs, points, circles, ellipse. 

I made a short video showing how to get in to the 2d sketcher and showing a dxf import.
https://youtu.be/_S8Bq3GKX04

I don't quite understand the title of your issue.

> performing operation with current parameters leads to an invalid object (self-intersecting / zero-thickness / complete degeneration or unsupported cases) 

Could you explain further the problem you are having?

mmiscool, 2023-03-19:
Closing issue @oguzhannkayaa 
Go ahead and reopen the issue if needed. 

To use the 2d sketcher goto https://hostLcation/sketcher.html rather than https://hostLcation/index.html

Only DXF importing is supported at the moment. The proprietary DWG format is not supported.

closed 2023-03-19 by mmiscool