xibyte/jsketcher #213: Integration with CADdrive?

ghackenberg, 2024-06-18:
Dear JSketcher community,

Love your project! ♥️ I was wondering if you are open for helping build an integration with [CADdrive](https://github.com/ghackenberg/caddrive)? I think, the two projects would fit nicely together. Looking forward to hear your thoughts on this...

Cheers!

ghackenberg, 2024-06-18:
**PS: Here is a screenshot of the CADdrive model revision graph.** I think, it would be awesome if the users can edit the CAD models directly within CADdrive using JSketcher. Each edit session then could lead to one or more revisions in the CADdrive model revision graph...

[image]

mmiscool, 2024-06-25:
Please react out via the contact form on https://autodrop3d.com so that we can exchange contact details and set up a meeting to discuss.